ThomasHeb/4AxisFoamCutter #4: Error when compiling for Arduino

ThomasHeb, 2020-09-09:
Hallo Thomas, bekomme beim Kompilieren einige Fehlermeldungen, könntest du mir da bitte weiterhelfen?
Danke!

Mehrere Bibliotheken wurden für "U8x8lib.h" gefunden
 Benutzt: C:\Users\Jetpilot\Documents\Arduino\libraries\U8x8libr
 Nicht benutzt: C:\Users\Peter\Documents\Arduino\libraries\Lib_8UG2
 Nicht benutzt: C:\Users\Peter\Documents\Arduino\libraries\U8g2
lcd.cpp:7:10: fatal error: SdFat.h: No such file or directory
 #include "SdFat.h"
          ^~~~~~~~~
compilation terminated.
exit status 1
SdFat.h: No such file or directory

ThomasHeb, 2020-09-09:
Bitte prüfe, dass nur eine U8G2 Lib installiert ist. 
Öffne hierzu in Arduino unter Werkzeuge/Bibliotheken verwalten...
Suche hier nach U8G2.
Prüfe ob mehrere Libs installiert sind.
Wenn ja, lösche diese in deinen Lib Verzeichnissen (C:\Users\Peter\Documents\Arduino\libraries\ bzw C:\Users\Jetpilot\Documents\Arduino\libraries), am besten löscht du alle U8G2 Libs
Installiere die U8G2 Lib neu (U8g2 by oliver Version 2.27.6 siehe Screenshot
https://github.com/olikraus/u8g2
[image]

Installiere die SDdFat - Adafruit Fork by Bill Greiman Version 1.2.3 Lib, siehe Screenshot
https://github.com/greiman/SdFat
[image]

Einbinden einer Lib in Arduino:
Lade die Lib als Zip-Datei von Github
[image]

Unter Arduino/Sketch/Bibliothek/.ZIP-Bibliothek hinzufügen... wählst du die ZIP Datei aus.
[image]

Die Libs erscheinen dann in den Arduino-Lib Verzeichnis
Hier musst du auch die U8x8lib.h and U8x8Lib.cpp (download von hier aus dem Projekt) im Verzeichnis U8G2 tauschen.
[image]

Campilot, 2020-09-10:
Danke Thomas!
Alle Fehler bis auf einen sind gefixt - hab aber noch immer den LCD Error

[image]

Campilot, 2020-09-10:
@ThomasHeb 
 U8x8lib.h and U8x8Lib.cpp hab ich natürlich ersetzt :)

ThomasHeb, 2020-09-10:
Hallo,

bei dem Stand "fatal error: lcd.h:  no such file...." fehlte wirklich die lcd.h und die lcd.cpp aus meinem Source-Code. Sieht man auch, wenn man deine beiden Screenshots vergleicht... da fehlt in der Arduino IDE die Lasche mit lcd.h und lcd.cpp.
Die Fehler bzw das Hängenbleiben hängt mit der Installation der U8G2 Lib zusammen.
Ich würde vorschlagen, dass Du einfach meinen Lib-Source 1:1 nimmst, liegt hier: https://github.com/ThomasHeb/4AxisFoamCutter/tree/master/03_Firmware/libraries . Nur die u8g2_font.c fehlt (Verzeichnis /libraries/U8g2/src/clib/), da > 25MB.

Schließe die Arduino IDE.
Bitte sichere die die u8g2_font.c aus deinem library Ordner. 
Lösche die Libraries in deinem Ordner
Kopiere die beiden Libs von hier rein. 
Füge die u8g2_font.c unter /libraries/U8g2/src/clib/ hinzu
Starte die Arduino IDE neu und kompiliere den Foamcutter.

ich wäre morgen Nachmittag auch gut zu erreichen.... glaube das ist nur ein banaler Fehler, der sich aber gut versteckt.

LG
Thomas

Campilot, 2020-09-11:
Hallo Thomas!

 

Zuerst mal vielen Dank für deine Unterstützung!!

Gute Nachricht, jetzt hat es funktioniert - anscheinend ist Oliver´s Lib korrupt?

 

Ich hab in den letzten Monaten 4 Foamcutter gebaut (hab auch viel Kontakt mit Keith Howlette) -  wenn du an der Mechanik interessiert bist, dann schick ich dir gerne einen Link zu meiner Dropbox - die meisten Teile sind 3d gedruckt und Antrieb über Riemen mit V-Slot Rollen!

 

Nochmal Danke und LG

Peter

ThomasHeb, 2020-09-11:
Hallo Peter,

freut mich, dass es jetzt klappt. Ich werde in den nächsten Tagen noch ein Update machen, um die Verfahrgeschwindigkeit auch lokal direkt an einer Taste umzuschalten. Du kannst Dir ja einen Notifier in GitHub legen, dann bekommst du eine Info, wenn ich die Firmware hochlade.
Gib mir auch gerne bescheid, wenn du weitere Funktionen brauchst.

An der Hardware bin ich auf jeden Fall interessiert.

Brauchst du die Libs noch (https://github.com/ThomasHeb/4AxisFoamCutter/tree/master/03_Firmware/libraries)?
Wenn nein, würde ich diese hier löschen, da ja nicht von mir geschrieben.

LG 
Thomas

Campilot, 2020-09-11:
Hallo Tom,

 

ja mich freut´s auch...aber so richtig :D

den Notifier check ich gleich!

Die Libs brauche ich nicht mehr, hab sie lokal bei mir am Rechner ;)

 

Ich kopiere bis Sonntag mal sämtliche Daten in einen Dropboxordner und schick dir dann die Einladung!

 

hier ein link zu meiner ersten Styro Cnc -> https://www.youtube.com/watch?v=LLoHfbii8cM&t=23s

da hab ich aber noch einige Dinge grundlegend verändert, speziell die Drahtführung (war bei V1 noch mit einer Feder gespannt)

 

LG

Peter

ThomasHeb, 2020-09-11:
Hallo Peter,
freut mich! Video habe ich geprüft, wenn du willst kann ich deine Mechanik / Videos auch gerne mit hier verlinken.
LG
Thomas

closed 2020-09-11 by ThomasHeb