ThomasHeb/4AxisFoamCutter #5: Slight confusion with modifications and connections to ramps1.4 board

goos766, 2020-09-23:
https://ibb.co/WPXVmNd I am not sure if I understood your connection description correctly. Therefore, I am sending you this diagram to confirm that it is correct. I would be grateful if you could point out the errors in the diagram. I know that I mixed up a bit in the connections to the 100nf capacitors at the ends stops, because I didn't read it carefully. One more question, what should 100nf capacitors look like, what should I really look for? From what I've read, I conclude that I should solder one 100nf capacitor for each of the four ends tops, connecting the signal and ground. Can you provide details so I don't make a mistake and burn the board when buying wrong parts, can you send a link?

goos766, 2020-09-23:
Is using an external MOSFET like in 3d printers completely incompatible with your firmware? The external MOSFET seems to retain the PWM functionality. I am asking you as an expert in this matter because I do not know much about it.

ThomasHeb, 2020-09-27:
First of all, sorry for my late reply. I know that is frustrating, if a project get's sucked.

Your wiring has some errors.
- Stepper drivers are assigned correct
- capacitors on limit switches are correct, wiring see picture below
[image]
use a very small 0805 100nF capacitor and solder it from the bottom side of the ramps, directly between the pins (picture is not soldiers, but shows the first position, in addition solder the next 3 headers to the right lower side a cap, too. one 100nF for each limit switch line Signal (S) to GND (-) ... 4 Cs in total).
[image]
- buttons X/Y/U/Z arrangement is wrong, X/Y is the left side, U/Z the right side. See attached picture
[image]
- Cut of pin 10 and connection to pin 7 is not correct, see attached picture. please cut of only the pinheaded on pin 10 on the ramps board. no modification on the Arduino. Wiring between the pins as shown in the picture
[image]
- power supply / hotwire / fan connections are correct
- button wiring:
  - stop, back, speed, hotwire buttons are correct (ramps labels D48, but it is D40 in the Arduino)
  - X+/-, U+/- wiring are correct (button arrangement is not correct, see above)
  - swap Z+ and Z-
  - swap Y+ and Y-
  - please check the FoamCutter.ino for wiring the buttons
You can use an external MOSFET, but it is not required. If you have a power supply with app. 28... 30V you will have a current of app 0.5A for 80cm of hotwire. This is less than the stepper drivers will need. so you can connect the hotwire directly to the ramps. If you want to use an external MOSFET, you can grab pin 8 on the arduino to control the MOSFET. pin 8 is active high. I would recommend to use internal MOSFET, or if not possible. control your hotwire manual, by switching it on before cutting and off after cutting (i.e. some chargers for RC cars/planes have a hotwire function).
[image]

You are the second guy asking for a windows version of the post processor. If someone can do the user interface, I could support with a common/universal class file for the post processing functions.... let's see what happens in future...

goos766, 2020-09-29:
To begin with, I want to thank you very much for your effort to write such a wonderful description. For now, I am waiting for a few more parts that have been sent to me from China. At the moment I am designing my version of the foam cutter based on the rckeith.co.uk design. I will try to deal with electronics soon.

[image]

goos766, 2020-09-30:
I made corrections to the wiring diagram, I hope I have not missed anything
[image]

goos766, 2020-09-30:
[image]
My miniature version of the foam cutter controller

ThomasHeb, 2020-10-01:
Hi,

wiring looks much better. I only found Z-/Z+ to be swapped: Z- goes to D4, Z+ to D11.
I use two LEDs on D65 (Hotwire) and D66 (Speed). This indicates if the wire is on or if I switched to seek speed (positioning menu). But they are not a must.... you will see it, if the machine goes fast and you will smell it, if the wire is on ;P

It is a very good idea to include the button on the display in your design, If you have any wishes (placing a function like homing on it,...), let me know.

Please be aware, that you select most of the menus by pressing the rotary button. This needs some force, so everything has to be stable and should not tilt. 

Overall this is an awesome design! Thanks for sharing.

goos766, 2020-10-02:
The case design is a cluster of models from thingiverse.com. As for the stability of the housing, I am going to add tabs for the screws. The button on the housing of the screen is the reset button, so I think I can select the homing function from the menu. If I have no problems with uploading the software. At the moment, I am still struggling with the mechanics of printing parts and selected the cheapest and best solutions, trying to maintain the functionality and rigidity of the structure.

ThomasHeb, 2020-10-03:
The tabs for screws are a very good idea. Maybe you can separate the green and the yellow button fields a bit and add a raised stabilizer between from top to bottom. Keep the two wires and connectors between displayboard and ramps in mind. I think it is very useful to have a connector between the electronics and the mechanics, so I added a sub-d connector to the housing (walls must be very stable in the area).

The outer design is very compact.... good job.

goos766, 2020-10-06:
[image]

goos766, 2020-10-07:
After uploading the firmware, I was surprised that the reset button on the screen does not work like on 3d printers. Now I know what you meant by saying that you could assign to this button a function other than its default function, which is reset.

ThomasHeb, 2020-10-07:
Hi Goos,

if you use the slicing menu, you select the cutting direction with the buttons. 
Or use the jog for setting new max position (machine must be homed and positioned correctly to the max cutting position), or do a back travel.

So slicing makes only sense with the buttons (and the steppers). limit switches are not needed for slicing.

You can quickly define the onboard button as back button inside led.cpp, change
#define BTN_ONBOARD                   0x00000001 
to
#define BTN_ONBOARD                   0x00000080

I would recommend to follow the next steps:
- connect the buttons
- connect one stepper driver and motor
- connect 12V DC to Ramps 3/4
- go to position menu and check if you can move this stepper with the -/+ buttons
this helps you to check, if buttons are wired correctly.

goos766, 2020-10-07:
I'm glad you answer me at all. I appreciate your willingness to help strangers. I am pleased with your every answer and your time devoted to dealing with my problems.

goos766, 2020-10-08:
A question from a series of unnecessary questions: where did you put the fan? If you used a fan connection only to cool Ramps 1.4 then why not keep it on all the time while the unit is running. You made a separate menu for the fan; does it mean that I should place the fan somewhere else? For example, to cool the hot wire after finishing cutting.

ThomasHeb, 2020-10-09:
Hi Goos,

category „afterwards you are smarter“:
I saw fans in some other videos, cooling the stepper driver and the powersupply for the hotwire. So I placed some fans, wrote a small interface to test them (fan menu), with the idea that you can control them later on via gcode and found out, that nothing is getting hot......

The foamcutter works fine with devfoam2. Keith gave me some files to test and all of them worked very good. You can use the gcode files on sd card or the usb interface as gcode sender. Check out the page of Keith for some tools. Search the web for foamcutter and excel..... in the end the gcode is very simple. If you want, I can send you some simple files from my post processor for testing.

Software or tools.... you are not limited to my solution. I am just a Mac only user, with no computer in my workshop. So I decided to ADD sd card to the grbl on the arduino - sending gcode via usb is still working. on the other hand I wanted to use SketchUp, so I wrote this tool and the post processor for MacOS. But you can use all other software like devFoam, Jedicut,.... gcode is simple and most of them have the gcode sender already included. 

If you want to cut simple shapes, like letters, you only need coordinates around the letters (this I do with Sketchup), add some offset and control commands for the machine (post processor). So you can go for a simple CAD program with Excel post processing.....

May I ask what type of shapes you want to cut?

goos766, 2020-10-09:
Initially, I was thinking about cutting letters and vector graphics, but who knows, maybe in the future, when I get to know the operation of the machine and become confident in what I am doing, I will start something more complicated. I prefer to leave myself this margin for expanding the possibilities over time, while keeping the same state of the machine, maybe with some minor modifications / improvements. As for the previous question, I must admit that it is very good news that I can choose the software, to be honest I started to be a bit concerned about the sense of the whole project. I chose your software version mainly because the machine is computer independent. I am not a fan of a connected (dedicated only for this purpose) computer to any type of CNC machine.

ThomasHeb, 2020-10-09:
I added an example for text with SketchUp / Foamcutter.rb, w/o post processor to GitHub. Check out the main section at the lower end and the video https://youtu.be/lMjuI5fZvUQ. Feel free for any further question and have a nice weekend.

goos766, 2020-10-09:
[image]
[image]
I'm making very slow progress and still don't have all the electronics parts. And 3d printing is not one of the fastest parts of this project, especially with the amount of plastic needed to obtain good stiffness of these parts.

ThomasHeb, 2020-10-10:
this looks awesome!

goos766, 2020-10-12:
This time, the theoretical question about the power supply: is it necessary 30 volt power supply? Do calculators that can be found on forums related to the topic of selecting hot wire have any positive reflection in reality? Calculators show that the number of amps is more important than the number of volts And what material the wire is made of to obtain the desired resistance at the given cutting distance. The wire thickness is calculated based on the available amps and volts of the power supply, but with Ramps 1.4 hot wire, we only have 11 amps available, I believe. I think I can choose the thickness and type of wire to power the entire device from a 12 volt 30 amp power supply, as I mentioned before using a MOSFET. At least in theory according to the calculator, even the 11 amp output from Ramp 1.4 is sufficient to heat a hot wire at 12 volts.

I'm not sure if that's all what I wrote makes sense.

ThomasHeb, 2020-10-12:
I was worried about the power required for the hotwire, too. I bought a proxxon hotwire from my local workshop to do some tests. I have a wire length of 85...90 cm. With 27....28V DC, power setting 90%, results in 0,6 .... 0,7 Amps. Normal styro (the white one), I can cut with 300 to 500 mm/min. XPS (the good stuff) is working with 100 mm/min.
So for my application, 30VDC power supply is perfect and a simple DC/DC converter supplies the arduino and the stepper with 12V at 0,5A to 1A max (measured on the 30V input.... so app 2A on the 12V side).

To heat up the wire, you need power (volt x amps). longer wire needs more power (linear relation). with the voltage you can control the amps (amps = voltage / wire resistance). wire resistance is related to diameter (thiner = more resistance), length (longer = more resistance, linear relation), material. 
You can control the hotwire with a constant current source (constant amps). But in the end this source must be able to deliver the voltage, too. and with the assumption, that the hotwire resistance is some how constant, you can use a much simpler voltage source to control the hotwire. Simplest voltage source is a power supply with a PWM MOSFET.

The proxxon wire has a relative high resistance and is good for short wire length. I prefer low temperature and slow feed speed for complex shapes. If your wire length is 1,6m you will need 60V. 

I would recommend some tests.
Use XPS foam.
Wire length is diagonal length of your machine + some cm.
Your powersupply should have less than 60 VDC output (safety).
Mark a distance of 100mm on the foam.
Heat up the wire and put the foam on the wire. the wire should cut the foam with minimal force and nearly no bending of the wire.
Test different wires.

Hotwire on amazon:
https://www.amazon.com/s?k=resistive+wire&__mk_de_DE=ÅMÅŽÕÑ&ref=nb_sb_noss_1

Proxxon Hotwire:
https://www.amazon.de/Proxxon-28080-Ersatzschneiddraht-Thermocut-Rolle/dp/B0016ODWWC/ref=sr_1_1?__mk_de_DE=ÅMÅŽÕÑ&dchild=1&keywords=proxxon+28080&qid=1602468872&s=diy&sr=1-1

XPS:
https://www.hornbach.de/shop/XPS-Hartschaumplatte-Precit-300-G-1250x600x30mm/5097832/artikel.html

goos766, 2020-10-12:
I checked with online calculators how to choose a power cable and found out I have to buy 14awg / 2 mm2, choosing a stainless steel cutting wire, which have lower resistance than, for example, kanthal wire. I wonder about the distance between the towers of about 60-70 cm and a wire thickness of 0.36 mm. At least this is what calculators say, so I can still only theoretically say that I am prepared.

ThomasHeb, 2020-10-13:
Hi Goos,

the calculation sounds strange.
If you reduce voltage, the current will reduce, too. this is physics (trust me, I studies electronics). stainless steel wire won‘t work, you need NiCr wire. My wire length is about 85...90 cm ( tower distance is 72 cm) and I use the Proxxon 0,2mm NiCr wire with 28V x 90% Powersetting x 0,7A = 18W. easy to control on the ramps! 
The Excel you are using is trying to keep the Power per inch on the wire constant (W). but this is only for choosing a wire, not for controlling it later on! in my equation above, you can replace the Amps by I = U / R, so Volts x Volts x Powersetting (0....100%) x Powersetting / total wire resistance = actual power in the wire.

On my machine, I get app 35 Ohm total wire resistance (28V x 28V x 90% x 90% / 18 W).
With a length of 90 cm, I have 0,39 Ohm/cm or app 1 Ohm/inch or 11,9 Ohm/feet

On your machine:
Total wire length is app 1,6m (120 x 65 x 65). with power per length as a constant...... you need 18W x 160cm / 90cm = 32W. 
assuming you take a 30VDC powersupply, running it on app. 28V. with a powersetting of 90% in the foamcutter (my configuration).... 28V x 28V x 90% x 90% / 32W = 19,85 Ohm total wire resistance. or 19,85 Ohm / 160cm = 0,12 Ohm/cm = 0,315 Ohm/inch =  3,66 Ohm/feet
this should work: https://www.amazon.com/Kanthal-A1-Gauge-Resistance-Wire/dp/B07CHYL39X/ref=pd_bxgy_img_2/144-9922268-4104130?_encoding=UTF8&pd_rd_i=B07CHYL39X&pd_rd_r=6c963cc2-a31d-4869-a382-a52f86ecd330&pd_rd_w=iaf9G&pd_rd_wg=S6LD3&pf_rd_p=ce6c479b-ef53-49a6-845b-bbbf35c28dd3&pf_rd_r=GWN8A698W3J3EYBCJQSK&psc=1&refRID=GWN8A698W3J3EYBCJQSK
maybe you can find a wire with 0,2mm thickness and a resistance of app 3,5 Ohm / feet.

Current required maximum (30V and Powersetting =100%): 30V/19,85Ohm = 1,5A.
The ramps can handle this...no fan is required. you do not need to cut pin 10.
Calculation with the recommended wire: total resistance 16,8 Ohm, max current 1,7A @ 30V. No problems.

wires from ramps to hotwire and powersupply to hotwire.
I use AWG12 because I had them here... AWG14 is fine.

You should continue with the project. everything looks fine and the details can be solved. If you want, I can support you with the electronics. I have a ramps, all cables, parts and the powersupplys here. Let me know and I can do the modifications, test the board and send it over to you and put some cables and connectors in the box, too. (as always I ordered 5x more than I needed).

goos766, 2020-10-13:
I am looking for the cheapest solution, but as a beginner in this matter I can not verify whether I should buy a [laboratory power ](https://www.aliexpress.com/item/1005001313647177.html?spm=a2g0o.productlist.0.0.415b77e6bopCdK&s=p&ad_pvid=202010131014132481026579523480001021222_3&algo_pvid=19a968e2-a554-44bb-ba84-7f56d263151d&algo_expid=19a968e2-a554-44bb-ba84-7f56d263151d-2&btsid=0b0a187916026092538634391ec91f&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_)supply, which, even on aliexpress, is a bit too expensive for my budget.

ThomasHeb, 2020-10-13:
Hi Goos,

a laboratory power supply is a good solution, if you will work on similar projects in future. for this project a compact supply works, too. you will need app. 2...2,5A, or 70...80W. check out the LRS-150F-24 or LRS-150F-36 (20...25 EUR) both can be adjusted to app. 28V and have 150W.  these are clearly marked how to connect. but never the less, get in contact with an electrical specialist to check right grounding and fusing and protection and ....., to ensure safety.

https://www.meanwell-web.com/en-gb/ac-dc-single-output-enclosed-power-supply-lrs--150f--24

toroidal transformers need additional electronic components to generate DC, so you will not save money nor time.

goos766, 2020-10-13:
So I don't need to look for something that is 10A or even 30A. Even 4.2A is enough? Should I treat this as a separate power supply for the cutting wire only?

goos766, 2020-10-13:
It turns out that this is my fault and my problem with superficial reading of the text. I have re-read everything you wrote and will try to buy the power supply you suggested. Sorry again for not reading carefully. Thank you for your effort in describing everything I ask in such detail, although sometimes I don't understand everything and it's not about the language but about the electrical stuff.

ThomasHeb, 2020-10-13:
Hi Goos,

you don‘t need to worry, theoretical it is really hard do get everything. I go downstairs and check it. 2...2,5A is the complete consumption, including hotwire. you should choose a bit bigger supply, so that you do not need any fans there.

complete setup looks like this.
100...250V AC powersupply to 28VDC for general supply and hotwire
DC/DC from 28VDC to 12VDV for stepper, ramps, ....

goos766, 2020-10-13:
I am grateful for proposals to send me equipment, but I must politely decline. I respect someone's money and I know that earning money is not easy. I would feel guilty accepting the presents knowing how much effort it costs. It seems to me that I am able to finance this project myself. Thanks again for this generous offer.

ThomasHeb, 2020-10-14:
Hi Goos,

my offer is more a exchange.... modified ramps for unmodified, just to help you to get the system running with all the electrical stuff.

ok, now I got your issue. But it is not about the MOSFET on the ramps 1.4/1.6. Its about the voltage regulator on the Arduino and the Stepper. So sorry for my new tipps & tricks..... a bit like crazy new ideas every day.

There is a linear regulator on the Arduino and with more than 12VDC (supplied over the ramps) it is getting overheated. And the stepper should be supplied with 12V, too. 
But for the hotwire I designed for 28V (in my design to keep the current low) or you reduce the wire resistance to app. 1,6 Ohm / feet.
Then you can supply the current from a single 12V source over ramps 1.6 (use hotwire with app 1,6 Ohm / feet)
Current will climb to 2,7 A for Hotwire and 1,4A for Arduino/Stepper/..., so total 4,5 A (My supply can only drive 3,5 A, that's why I went for the low current solution). 

You can use the LRS-150F-12.
https://www.meanwell-web.com/en-gb/ac-dc-single-output-enclosed-power-supply-lrs--150f--12?returnurl=%2fen-gb%2fsearch%3fq%3dlrs-150f-12%23lrs-150f-12
you do not need a DC/DC Converter in this case.

To your question of 28V on the "11A" input. The input is separated from the rest of the circuit on the ramps 1.4 (not on the ramps 1.6), only a common ground. the MOSFET just closes the the hotwire to ground (inner circuit: 28V Input - fuse - connector hotwire - connector hotwire - MOSFET - ground), the MOSFET itself can manage more than 50V and has a very low inner resistance. So low current over the MOSFET means cool MOSFET. but even with 2,7A you are far away from overheating this MOSFET. And ramps 1.6 has a cooling mounted on the MOSFET. Only an onboard LED is in that circuit of "11A" but the preresistor with 1k8 is high enough to prevent any damage.

Cutting and rewiring Pin 10 is only for the fan. you don't need that fan. build the fan in your enclosure but do not wire/connect it. Leave the pin 10 as it is. I do a test with a unmodified ramps 1.4 to check if it is working without the fan and if not, you get a modified firmware.
This will work with ramps 1.6, too, when you use the 1,6 Ohm/feet hotwire and a 12V power supply.

goos766, 2020-10-14:
More resistance means thinner cutting wire, or I get it wrong again. I currently have a 12v 30A power supply (at least this is what it says on the sticker), which in the past I was supposed to use for a 3d printer in the ramps 1.4 configuration in case it broke something inadvertently and I needed a quick replacement, but it did not happen so I have a new one / unused.

For now, I am trying to solder cables to use parts lying around unused for several years. The key idea here is to connect DuPont plugs with JST plugs, it could be done probably cheaper using a crimper, but in recent years I have bought a few crimping tools and never suitable for such small things. I have some female to female DuPont cables for which I did not find any use and some JST cables left after the ender 3 cabling modifications. I figured I wouldn't be spending my money again on something that would come in about 2-3 months if at all so I cut a few cables and solder them now.

ThomasHeb, 2020-10-14:
Hi Goos,

more resistance means thinner cutting wire, correct.... but if you go from 28V to 12V you need less resistance, to get the required power on the wire (32W = 28V x 1,4 A = 12V x 2,7A). This may lead to thicker cutting wire. My 12V recommendation is 
https://www.amazon.com/-/de/dp/B07CHX4MMT/ref=psdc_495310_t2_B07CHSG169
Diameter is 0,64mm..... I remember, that's why I have chosen the proton with 0,2mm and played with the voltage, too.

I cutted all connectors, too and I am using standard pin connectors 2,54mm male/female as connectors. Glue them together and where needed some isolation ;)

I finished the test with the unmodified Ramps 1.4 board. Standard firmware without modifications:
You do not need to cut Pin 10 as long as you don't use a fan. Appending some screenshots. You see that the internal MOSFET stays very cool (10 minutes, ambient temp. 18°C, no fans active). So Ramp 1.4 and Ramp 1.6 will work. I would only recommend to add the Cs for EMC compensation.

28,5V / 0,9A: stepper + Hotwire
28,5V / 0,38A: stepper (10,83W)
Values are a bit different because of different ramps/arduino and because I did not do all parameter settings.

Less Voltage means higher current, because power consumption stays almost constant
25,6V / 0,42A: stepper (10,75W) ... only hotwire is not getting hot enough if  reduce more.

 
[image]

goos766, 2020-10-14:
ok. I took a step forward with all the stepper motors and their directions checked. Tomorrow and maybe even today endstops soldering. later ... ehh, what the hell, I will cut this pin 10, because without this fan I feel somehow insecure. I have to prepare for testing the cutting wire, because I do not have a power cable of the right length for this whole test to make sense.

ThomasHeb, 2020-10-15:
Hi Goos,

the resistance of the hotwire is measured directly on the hotwire. Between the points, where you connect the cables.... see picture. 
[image]
[image]

Resistance is the base for choosing the right wire.
Connect everything. Go to the hotwire menu , select manual and start with a low power setting.
Put some foam on it... the foam should not be cut.
increase power setting until the foam is cut without putting fore on it. this is the setting for slow cutting speed (app 100 mm/min). Check out the video attached in the zip.
Put this setting into the parameter ($5 = feed speed and $27 = hotwire power is stored automatically from your last manual setting).

If your wire is not getting hot enough:
- increase input voltage (only with ramps 1.4: 28V >>> 30V)
- reduce wire length (put the crocodile connections closer together
- chose a wire with a lower resistance .... thicker wire ;(

If your wire is getting much too hot / cutting is starting below 50% powersetting:
- there is no urgent need to change something
- increase wire length
- choose a wire with higher resistance >>> smaller diameter ;)

I keep my fingers crossed for your fist cut!

[IMG_4588.mov.zip](https://github.com/ThomasHeb/4AxisFoamCutter/files/5382446/IMG_4588.mov.zip)

goos766, 2020-10-15:
So it seemed to me at the beginning. But after reading that the power cable also influences the resistance of the entire installation, I started to doubt a little where this resistance measurement should actually take place. Thanks for that clarification.

ThomasHeb, 2020-10-15:
Of course everything has an influence. But to get that baby working, you can focus on the major topics ;o)
Cables have less than 1% impact compared to the hotwire/resistive wire.

closed 2020-10-21 by goos766