ThomasHeb/4AxisFoamCutter #7: It seems I am doing something wrong

goos766, 2020-10-07:
[errors.txt](https://github.com/ThomasHeb/4AxisFoamCutter/files/5337663/errors.txt)

goos766, 2020-10-07:
I tried to follow your description but it seems like I mixed up something. The Arduino IDE libraries for Windows are located in two places. Again, I don't understand what I'm doing wrong. I have replaced the U8G2 files in the scr folder. I have updated / downloaded the libraries you mentioned in the description. Then I realized there were two library locations so I swapped files in both folders. I guess somewhere I had to do more than I needed. I sent you the error list that the Arduino IDE generated. I'm not sure, but I think someone had a similar problem, only the conversation was conducted in German.

goos766, 2020-10-07:
[errors2.txt](https://github.com/ThomasHeb/4AxisFoamCutter/files/5337838/errors2.txt)
I tried again and got a similar result on a freshly installed Arduino IDE. With the procedure you recommend

goos766, 2020-10-07:
avrdude: stk500v2_receivemessage(): timeout avrdude: ser_send(): write error: sorry no info avail avrdude: stk500_send(): failed to send command to serial port avrdude: ser_recv(): read error: the handle is invalid.

I found this thread in German that I mentioned earlier. The file verification was successful, but unfortunately I got the above error at the end.

goos766, 2020-10-07:
[http://www.wch.cn/download/CH341SER_EXE.html](url) This file solved my usb 3.0 communication problems on Windows 10 with the Chinese Arduino Mega board. This is much more of a complication than with the 3D printer setup, and it would seem that the device is much simpler.

closed 2020-10-07 by goos766

ThomasHeb, 2020-10-07:
Hi Goos!

sorry for my late reply... seems that we are working in different time zones ;o)

It seems, that you are linking the arduino.h. This header is a standard managing all background tasks.... and this is not compatible with the grbl. I think the SdFat brings the arduino.h back to the linker. Your errors are showing some linked files that are not included in the version I used.
Please use SdFat by Bill Greiman, **Version 1.2.3.**

Or use this libs https://github.com/ThomasHeb/4AxisFoamCutter/tree/master/03_Firmware/libraries

Only the file u8g2_font.c is missing (path /libraries/U8g2/src/clib/), due to > 25MB.

close the Arduino IDE.
save u8g2_font.c from your lib folder
delete the libs in your lib folder
copy the libs from the link above.
add u8g2_font.c under /libraries/U8g2/src/clib/
Start Arduino IDE and compile Foamcutter.