ThomasHeb/4AxisFoamCutter #8: Code wont compile "'gc_command' was not declared in this scope"

LarsBrinck, 2020-10-11:
Hi Thomas,
Thanks for your work with the foamcutter! I´ve been looking for a foamcutter that I can run from my mac and when I finally found your brilliant standalone solution I got around to built myself a small one like this: https://www.thingiverse.com/thing:3676825 

Unfortunately I ran into troubles trying to compile the code to my Arduino. I guess there is some library or something missing. I´ve read the other issues here and tried to re-load the libraries as you suggested but I get the same error.
[Error Foamcutter Lars Brinck.txt](https://github.com/ThomasHeb/4AxisFoamCutter/files/5361304/Error.Foamcutter.Lars.Brinck.txt)

Can you see what is wrong or do you need more info?

ThomasHeb, 2020-10-11:
Hello Lars,

there are many errors in the log and I guess some are just based on the others. There is a redefinition of some macros like max and min, this looks like the arduino.h is linked. I think the SdFat brings the arduino.h back to the linker. Your errors are showing some linked files that are not included in the version I used.
Please use SdFat by Bill Greiman, Version 1.2.3.

Or use this libs https://github.com/ThomasHeb/4AxisFoamCutter/tree/master/03_Firmware/libraries

Only the file u8g2_font.c is missing (path /libraries/U8g2/src/clib/), due to > 25MB.

close the Arduino IDE.
save u8g2_font.c from your lib folder
delete the libs in your lib folder
copy the libs from the link above.
add u8g2_font.c under /libraries/U8g2/src/clib/
Start Arduino IDE and compile Foamcutter.

ThomasHeb, 2020-10-13:
Hi Lars,

good job and thank you for the feedback. I will integrate and correct this in the next changes. Let me know, if you need any further support.

LarsBrinck, 2020-10-13:
Thanks Thomas. I´ll close this issue and open a new if I get stuck again.

closed 2020-10-13 by LarsBrinck