ThomasHeb/4AxisFoamCutter #9: SketchUp Foamcutter.rb does not generate GCODE as shown

ThomasHeb, 2020-10-14:
From youtube.com

After I select the paths, I am not getting the GCODE window to pop up like yours does at 5:49  Any idea why???

Please send me your SketchUp file for a deeper analysis.

Regards 
Thomas

CWO4, 2020-10-14:
Here is the file requested.  Very simple cut going from square to round.  It is on the same SketchUp file you included as an example.  And also, the "Tower" gcode you included is not running correctly either.  The code actually runs and cuts, but the shapes are all wrong.

[Tower.zip](https://github.com/ThomasHeb/4AxisFoamCutter/files/5380110/Tower.zip)

CWO4, 2020-10-14:
Here is a photo of the shape that results from the gcode that is produced.

[image]

ThomasHeb, 2020-10-14:
thank you. I will check the code on my machine tomorrow.
was your machine already parametrized and did you made some successful cuts before?

the cut out looks a bit like axis-traveling for x and y is not ok and positioning of the foam is not correct. but this can also be an issue due to missing post process (square to round is a complex shape and needs a post process to calculate axis traveling from block shape, I will include some pictures tomorrow for better understanding)

CWO4, 2020-10-14:
Yes, my machine is working perfectly.  I built the one posted by RCKeith on
YouTube (Keith Howlett) and it has been producing perfect wings.  I just
wanted a way to get fuselage sections cut and I really like SketchUp.
SketchUp can generate .stl files for my 3D printing, but no GCODE.

Tom

ThomasHeb, 2020-10-15:
Hi Tom,
using SketchUp for fuselage is exactly what I do.
Your design is correct, you only missed the post process. Sketchup/Foamcutter.rb exports the two paths (left and right side) of the shape. If you do not run a post process, the foamcutter takes these patches as the machine movement. see picture below. if you add a foam somewhere in the middle, you get something between square and round.
[image]
What is a bit strange on your photos.... the width and hight of the cutout looks different..... these should be equal. 
I made a small video showing the steps and insides.
Actually this doesn't help you, because my post processor is running on MacOS. But what I can offer is to extend the foamcutter.rb-tool with a basic post processor... but this will take some time and testing
[postprocess.mov.zip](https://github.com/ThomasHeb/4AxisFoamCutter/files/5382605/postprocess.mov.zip)

[image]
[image]

CWO4, 2020-10-15:
Thanks for the clarification.

I am very sorry this can't be used on a Windows machine.  Too bad the
software is not Java based (or other cross platform capable software) as
there are a LOT of Windows users out there that would actually pay you for
a program to do this since SketchUp has no other plugins that will do the
job.

I will have to look elsewhere I guess.

Thanks again,

Tom

PS:  You should add a note on your YouTube videos that this only works on a
Mac.  Will save a lot of wasted time for people using Windows.

ThomasHeb, 2020-10-15:
Tom,
stay tuned. the first SketchUp post processor is working.... I will post a demo here and hope you can support me for testing.

P.S.: Maybe someone could use my source and port it to Java.

ThomasHeb, 2020-10-15:
Hi Tom,

appending you find the foamcutter_V002.rb file. Please open SketchUp, the Ruby Code Editor and load this rb file and execute it. you will find a new menu "Foam Cutter Tool PP"
[image]

I modified your example (attached tower.skp.zip).
you have to add your machine dimensions at (1) and (4). the distance in between is the the distance between your axis (I added my machine dimensions).
The shape is placed in between and is equals the position of the foam inside the machine and the thickness of the foam (distance between (2) and (3)).
[image]

Select the Foamcutter Tool PP. Select
- (1) left machine side
- (2) left shape side (be careful to use the correct line)
- (3) right shape side (be careful to use the correct line)
- (4) richt machine side

The tool calculates the movement of the axis, draws the movements inside SketchUp. exports the Gcode, including hotwire on/of, power setting and feed sped (see foamcutter settings menu).
[image]
As you can see, my machine is too small to produce this shape.... I would need to reduce the machine width.

[default.gcode.zip](https://github.com/ThomasHeb/4AxisFoamCutter/files/5387030/default.gcode.zip)
[foamcutter_V002.rb.zip](https://github.com/ThomasHeb/4AxisFoamCutter/files/5387031/foamcutter_V002.rb.zip)
[Tower.skp.zip](https://github.com/ThomasHeb/4AxisFoamCutter/files/5387032/Tower.skp.zip)

Can you please test this... of course you need to modify to your machine dimensions.

Regards
Thomas

CWO4, 2020-10-15:
That looks REALLY nice!  I will test it tonight.

Tom

CWO4, 2020-10-15:
I am enclosing the SketchUp file modified to show my part and my machine's
dimensions.

Part placement is apparently critical.  However, I did manage to gee a good
part after a couple of tries.

This was only my first attempt tonight.  I will try more tomorrow.

Thanks again,

Tom

ThomasHeb, 2020-10-16:
Hi Tom,

Thanks for the first feedback. Not everything is possible inside a simple script compared to 3D calculation in Swift, but if you have ideas and wishes, let me know!

I had several problems on my fuselage, too. the small segments have a width of 60mm and behind the wing a big difference between the left and the right shape paths, so my axis movements have been too small. I reduced operational width of the machine from 721 to 350mm and for the nose to 200mm, so I could use full axis movement to produce big angles between the axis. The nose had 2 outer and two inner cut-outs per side, which I combined in my post processor and checked visual, if the hotwire is not intersecting the final piece.

[image]
[image]

I tried to rebuild that in the SketchUp PP Tool and this is not possible in a single calculation. But you can calculate the 4 cutting steps separate and then copy the relevant parts from the gcode together to one big file... 

[image]

ThomasHeb, 2021-02-03:
foamcutter.rb with integrated post processor is released

closed 2021-02-03 by ThomasHeb