ThomasHeb/4AxisFoamCutter #10: Code is not running

LarsBrinck, 2020-10-16:
Hi Thomas,
AFter succesful uploading...nothing happens. Having verified the hardware setup using the rrd-glcd-test (found here on github)  I´m quite lost. Screen background lights up but nothing happens. I can´t communicate through the serial monitor (?,§,! etc) - no response. I´ve tried to insert some Serial.print lines, buzzer etc in the code to get some idea of where the code stucks but no luck.

Any idea of how to debug or to pin down the problem?

ThomasHeb, 2020-10-16:
Hi Lars,

the board must open the display and show a welcome screen and wait for a button/jog... or go to an error screen (limit switch or estop).... but it must show something and communication must work in the welcome screen.... i tested this two days ago with arduino / ramps / display and no other hardware and no modifications on the ramps.
What board/arduino do you have?
What size of code did you have?
ROM 65586 Byte / 25%, RAM 5788 / 70%

I am a bit worried, that still some conflict with the libraries exist...

Appending a hex file, check out xloader for direct download
https://www.hobbytronics.co.uk/arduino-xloader

Regards
Thomas

[FoamCutter.hex.zip](https://github.com/ThomasHeb/4AxisFoamCutter/files/5392270/FoamCutter.hex.zip)

LarsBrinck, 2020-10-16:
I´m off to work right now but can test this tomorrow night or the day after (sun). I have one cheap chinese arduino copy (12MHz!?) and one (expensive) original 16MHz arduino - I´ve tested both with same result.

I´ll be back as soon as I´ve tested.

Thanks for your help!
/Lars

ThomasHeb, 2020-10-16:
Hi Lars,

do you have some programming experience on arduino? Just asking from where I should start the test scripts.
which type of ramps are you using? a had no problems with the one I bought first, the second one did not work at all, until I found a cold soldering connection on a 12V pin from ramps to arduino.

Regards 
Thomas

LarsBrinck, 2020-10-17:
I got some basic experience doing a midi controller on a teensy making code snippets to make endless encoders behave ok, but that´s all. In this project I have called in support from a more experienced friend :)

Code size: "Sketch uses 67348 bytes (26%) of program storage space. Maximum is 253952 bytes.
Global variables use 6071 bytes (74%) of dynamic memory, leaving 2121 bytes for local variables. Maximum is 8192 bytes."

I can´t see any manufacturing data on the ramps board - I bought it from banggood so it is probably a chinese copy. But if I upload the code to the arduino without the ramps board connected, I should be able to communicate with it through the serial monitor using grbl commands -or not?

Seems to me as the arduino gets "bricked"...

Regs,
/Lars

ThomasHeb, 2020-10-17:
Hi Lars,

yes, good idea. Disconnect the Ramps. 
If your download is working without any error, you should be able to communicate.... 

ok, let's start from the beginning:
- open Arduino IDE
- goto File/Example/01. Basics/Blink
- a new ion file is opened... select this window.
- open menu tools and select board and processor Mega 2560
- select your com port
- open menu sketch/upload ... takes a second and you should get a "Done uploading."
- now the LED L should blink 1s on, 1s off

- goto File/Example/01. Basics/DigitalReadSerial
- a new ion file is opened... select this window.
// - open menu tools and select board and processor Mega 2560
// - select your com port
- open menu sketch/upload ... takes a second and you should get a "Done uploading."
- open menu tools/serial monitor. select on the lower right corner 9600 baud
- you should see many "0"s scrolling downward.
- close that window

- open foamcutter.ino. 
- open menu sketch/upload ... takes some second and you should get a "Done uploading."
// - open menu tools and select board and processor Mega 2560
// - select your com port
- open menu sketch/upload ... takes a second and you should get a "Done uploading."

- open foamcutter.ino. 
- in config.h set
   #define USE_LIMIT_SWITCHES                1  
   #define USE_BUTTONS                       1  
- open menu sketch/upload ... takes some second and you should get a "Done uploading."
// - open menu tools and select board and processor Mega 2560
// - select your com port
- open menu sketch/upload ... takes a second and you should get a "Done uploading."
- open menu tools/serial monitor. select on the lower right corner 115200 baud
- press the reset button on the Arduino
- both LEDs on the Arduino should be on
- you should see in the serial monitor
   Foam Cutter: Mega2560 with 4 axes (X, Y, U, Z)
   Grbl 0.8c2 ['$' for help]
- type "$" in the input Boxand press Send, you should see the help
[image]
- close that window

If possible download the hex file from my last post to your arduino with loader

Regards 
Thomas

fredrik-jacobsson, 2020-10-17:
Hi Thomas - thanks for your input!

Success with your hex-file and the XLoader. Both Arduinos run the code and I can communicate through the serial interface. Both Arduinos work with my Ramps board as far as the display is concerned. I haven´t tested with drivers and Steppers yet. (The endless encoder code really seems to be in need of rewriting but that is NOT the concern of this issue - I´ll be happy to see if my code can do the job later on:))

I had tried Blink and some other sketches to test the cards as a first check but it´s good to be sure :)

Still no luck uploading the code from my computers. (I´ve borrowed a PC since I get a avrdude:timeout() when I upload it from my Mac.) That code doesn´t work.

My guess is that you are right about the libraries and that I have conflicting libraries in the code I generate in my IDE. I switched over to this PC so the two imported Libraries are:
U8G2 ver. 2.27.6
SdFat_- Adafruit_Fork ver. 1.2.3

Perhaps it would be easier for me to start soldering buttons and install limit switches and run from your hex-file - but it seems so weird not to be able to see the actual uploaded code?

Any ideas where to go from here?

CHeers!
/Lars

LarsBrinck, 2020-10-17:
Thx for your input!

Success uploading the .hex file to both Arduinos. Grbl commands show up and both Arduinos makes the display run through my Ramps board so that seems to be working so far. (The endless encoder code needs rewriting but that is NOT the issue now. Lets see what we can do about that later:))

 I tried to upload code from the IDE of a PC that I have borrowed since my Mac gets this avrdude:timeout() error. No luck. That compiled code does not work. I think your guess that there are some conflicting libraries are right. This PC has U8G2 ver 2.27.6 installed and SdFat - Adafruit Fork ver 1.2.3 installed

Perhaps it would be easier for me to start soldering buttons and limit switches - but it seems strange not to see the actual code running?

Where do we go from here? I´ll try to attach drivers and steppers later tonight.

CHeers!
/Lars

ThomasHeb, 2020-10-18:
These are good news!
Nevertheless PC compiled source should work, too. Libs are correct.... so very strange....

Next steps are:
- add the buttons and LEDs to the Ramps. You can test them in the menu position
- calculate the steps per mm and set the microstepps according with the jumpers. Are you using belts?
- add the stepper driver to the Ramps.... maybe you set jumper over the limit switch pins on the ramps.
- adjust the driver https://www.makerguides.com/a4988-stepper-motor-driver-arduino-tutorial/ simply plug the stepperdriver in and measure/adjust the voltage over the resistor. 
- connect 12V DC to Ramps 3/4
- add a stepper motor and test operation

Regards
Thomas

LarsBrinck, 2020-10-20:
Hi Thomas,
I thought I would make another attempt to compile the code on the PC. Two questions arise:
1 - In your folder "03_Firmware" there are the "libraries" folder with the two libraries "SdFat_-_Adafruit_Fork" and "U8g2". Apart from that (and the folder "Foamcutter")there is a folder named "Lib_8UG2" - is that folder or the files within needed anywhere?
2 - You mention in your  instructions that the "fonts.cpp"-file is missing in your U8G2 library and should be copy/pasted from another U8G2 library. But the "fonts.cpp"-file is not missing in your library? Is this of any significance i.e. should I download the U8G2a library and copy that "fonts.cpp"-file and replace the one in your U8G2 library?

Sorry for the confusion but I´m just trying to find out why your .hex file works and not your un-compiled files...

(I really don´t want to solder buttons...:))

/Lars

ThomasHeb, 2020-10-20:
Hi Lars,

in the folder 03_Firmware is
- Foamcutter: The Arduino source code
- Lib_8UG2: these two files are modified inside the 8UG2 Lib and need to replace the original ones
- library holds the libs (all files) from my Mac. Another guy got some problems with the linker, too, so I overed my versions of the libs. Only the file u8g2_fonts.c is missing on my git, because of file size. please copy this from your libsfolder. delete your libs folder, paste the one from here and add the u8g2_fonts.c

[image]

I think you want to wirk with a version w/o buttons. appending the hex file 
[Archiv.zip](https://github.com/ThomasHeb/4AxisFoamCutter/files/5410125/Archiv.zip)

LarsBrinck, 2020-10-20:
Hi again,
Made an attempt on the PC and the results were not so uplifting:
With your U8G2 library it won´t compile - seems to be a font missing. With latest U8G2 library it compiles. And it compiles with both SdFat-Adafruit fork libraries, yours and 1.2.3.

BUT
I get the following errors in the compiling window:

[image]

and the arduino gets bricked.

SO
Would you like to compile a .hex file with buttons and limit switches set to 0 and I will try that?

OR
Just tell me to start soldering those (/&%€%& buttons :):):)

/Lars

ThomasHeb, 2020-10-20:
Hi Lars,

here we go.

[Limit0_Button0.zip](https://github.com/ThomasHeb/4AxisFoamCutter/files/5410142/Limit0_Button0.zip)
[Limit1_Button0.zip](https://github.com/ThomasHeb/4AxisFoamCutter/files/5410143/Limit1_Button0.zip)

the error above may result from the missing u8g2_fonts.c - see above.

LarsBrinck, 2020-10-20:
YES! It accepted the compiled code when I changed those two U8G2-files.
Plus - I can now change limits and buttons and it still compiles and the code is accepted.
When I set buttons to 0 the endless encoder started to work with the LCD screen.
Happy Happy!
Time to connect the steppers!
Thanks for your support!

ThomasHeb, 2020-10-21:
You are welcome!
Have fun with the project and happy cutting

closed 2020-10-21 by ThomasHeb