ThomasHeb/4AxisFoamCutter #12: Endless encoder fix?

LarsBrinck, 2020-10-22:
Hi Thomas,
I´m cutting with success! Very happy. But the endless encoder sucks...:) - but I can manage. But I used a small script for fixing this in another project (midicontroller to Lightning Software where I used a bunch of endless encoders.) I tried this script as it is on the reprap screen encoder and it works flawlessly. But I can´t implement it in the lcd.cpp sketch - my friend and I spent the afternoon thinking it would be easy but we just brick the encoder...

Try this script. If you could implement the script in lcd.cpp to get the encoder work properly I would be most happy. Again.

Still having fun,
/Lars
[Encoder_Test.ino.zip](https://github.com/ThomasHeb/4AxisFoamCutter/files/5424056/Encoder_Test.ino.zip)

ThomasHeb, 2020-10-23:
Hi Lars,
thanks for the feedback. I need to check that out over the weekend. The gcode and motion controller is controlling the loop timing. due to that the encoding is a bit tricky. I hope that I find some time to adapt your hints over the weekend.

Regards 
Thomas

ThomasHeb, 2020-10-24:
Hi Lars,

the Encoder_Test.ino didn't work at all, but the Rotary Library you linked is doing a perfect job! I merged this with the lcd.cpp. Could you please do some tests before I upload it for public use?

Please copy the library Rotary to your Library folder and use the attached lcd.cpp
[image]

[Archiv.zip](https://github.com/ThomasHeb/4AxisFoamCutter/files/5432620/Archiv.zip)

LarsBrinck, 2020-10-25:
Hi Thomas,
Works like a charm! Well done! I will now leave you alone for a while and start cutting some cores :) The cutter has all the functionality I need now - maybe I will attach the limit switches and a fan later on. BTW - there is a free 4-axis gcode program for wing profiles here: https://www.rcgroups.com/forums/showthread.php?3467209-WingWire-4-axis-gcode-freeware (by the same guy who shared the cutter I built on Thinggiverse). Again - BIG THANK YOU!!!

ThomasHeb, 2020-10-25:
Hi Lars,

you are welcome. Have a lot of fun with your cutter!

Regards,
Thomas

closed 2020-10-25 by ThomasHeb