ThomasHeb/4AxisFoamCutter #14: Compiled Firmware not working

ThomasHeb, 2020-12-05:
Request from youtube / Ugur Demirkan
https://www.youtube.com/watch?v=fht_X7mQ-qg&lc=UgyDhqStnxrQoc5TaMB4AaABAg.9Gsrt8Y3kGy9Gt05Y3vglM

Ugur:
I compiled the program I loaded it I put it in Lib_8UG2 library, but there is no screen, where is I making a mistake thanks

Thomas:
Hi Ugur, did you follow the steps I posted on github? which hardware are you using exactly (a picture would be great, especially showing the wires between the Ramps and the display)? have you tested the display with another program? If you agree, I would open an issue on github and try to figure out a solution? Regards Thomas

Ugur:
I have installed Repetier-Firmware and it works fine. But I am loading your firmware, there is nothing on the screen.
RAMPS 1.4 Mega 2560 LCD 12864 kit

Thomas:
many times there are bad soldered pins on the boards. but if the repetier firmware was running on the arduino/ramps/display, we need to go a bit deeper. I will open a github issue and start with some simple test programs. do you have experience with programming / writing code for arduino?

ThomasHeb, 2020-12-05:
Hi Ugur,

We start step by step, to ensure alle parts of hardware are working correct, all settings in the arduino IDE are good .... so maybe many steps seems to be annoying ... but it is much easier to start from a working point, than to track an error backward.

Test 1: Please open the Arduino IDE.
Set Arduino Mega and you Port in the Tools menu.
[image]

Open the Blink Example
[image]

Compile and Download to Arduino Mega (without Ramps Board)
you should see the LED closer to the power connection blinking
[image]
Is test 1 working?

Test 2: Disconnect from the Arduino from the USB and connect the Ramps board (no external Power is required, no stepper or display needs to be connected.... but if they are connected, there is no problem)
Reconnect to USB Port.
[image]
you should see the LED on the Ramps close to the Mosfets blinking.
Is test 2 working?

Test 3: Open Arduino IDE. Check Port in Tools Menu and compile and download the Blink Example. 
you should get no errors and the LED should blink again. This shows, that the Ramps is not disturbing the USB communication. Working?

ugurdem19, 2020-12-05:
[image]
I don't think there is a problem with arduino and lcd, Repetier-Firmware works fine. but 4AxisFoamCutter-master is compiling firmware but it does not work. The screen is white.

ThomasHeb, 2020-12-05:
Hi Ugur,

if the screen is white (filled pixels) this indicates an issue with the lcd driver.

please do the test 3, so that the blink is running on the arduino.
connect the lcd display and power over usb. the backlight should be on, but all pixels should be off (not white). is that working?

ThomasHeb, 2020-12-05:
Download the source from GitHub. you should get something like this
[image]

copy from 4AxisFoamCutter-master/03_Firmware/libraries
SdFat_-_Adafruit_Fork and U8g2 to the libraries folder of your arduino installation

open the attached LCD_Test.ino, compile and run it. you should see a text blinking
[LCD_Test.zip](https://github.com/ThomasHeb/4AxisFoamCutter/files/5647078/LCD_Test.zip)
[image]

ugurdem19, 2020-12-05:
[image]
no proplems, blink foam cutter

ThomasHeb, 2020-12-05:
yeah!!!!
ok. let's do the next step.
open the FoamCutter.ino from here, compile it and run it. you should see a welcome screen
[image]

[image]

ugurdem19, 2020-12-05:
Where will I download this file? Can you give a link?

ThomasHeb, 2020-12-05:
FoumCutter.ino is the source code for Arduino. you find it in the download from GitHub in the folder
4AxisFoamCutter-master/03_Firmware/FoamCutter.
you can open it directly from this folder.
[image]

ugurdem19, 2020-12-05:
https://github.com/ThomasHeb/4AxisFoamCutter/tree/master/03_Firmware I downloaded this file but it doesn't work

ThomasHeb, 2020-12-05:
please compile it again. do you get any warnings or errors, does the download work?

ugurdem19, 2020-12-05:
[image]
[image]

It is downloading, installing but not working.

ThomasHeb, 2020-12-05:
this is strange. 
have you deleted any old copy of the U8g2 from you arduino/libraries folder and only copied the one from her to that folder?

Another test is to upload a hex file directly.
Appending a hex file, check out xloader for direct download
https://www.hobbytronics.co.uk/arduino-xloader
[FoamCutter.hex.zip](https://github.com/ThomasHeb/4AxisFoamCutter/files/5647164/FoamCutter.hex.zip)

ugurdem19, 2020-12-05:
[image]

thank you very much for your help when i upload hex

ugurdem19, 2020-12-05:
After installing hexi, I noticed that it was giving the following errors but still loading
Arduino:1.8.12 (Windows 8.1), Kart:"Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from C:\Users\www\Desktop\4AxisFoamCutter-master\03_Firmware\FoamCutter\FoamCutter.ino:34:0:

sketch\nuts_bolts.h:30:0: warning: "max" redefined

 #define max(a,b) (((a) > (b)) ? (a) : (b))

 

In file included from sketch\FoamCutter.ino.cpp:1:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:93:0: note: this is the location of the previous definition

 #define max(a,b) ((a)>(b)?(a):(b))

 

In file included from C:\Users\www\Desktop\4AxisFoamCutter-master\03_Firmware\FoamCutter\FoamCutter.ino:34:0:

sketch\nuts_bolts.h:31:0: warning: "min" redefined

 #define min(a,b) (((a) < (b)) ? (a) : (b))

 

In file included from sketch\FoamCutter.ino.cpp:1:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:92:0: note: this is the location of the previous definition

 #define min(a,b) ((a)<(b)?(a):(b))

 

In file included from C:\Users\www\Desktop\4AxisFoamCutter-master\03_Firmware\FoamCutter\FoamCutter.ino:34:0:

sketch\nuts_bolts.h:34:0: warning: "bit" redefined

 #define bit(n) (1 << n)

 

In file included from sketch\FoamCutter.ino.cpp:1:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:124:0: note: this is the location of the previous definition

 #define bit(b) (1UL << (b))

 

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src/SPI.h:17:0,

                 from sketch\limits.cpp:14:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:92:0: warning: "min" redefined

 #define min(a,b) ((a)<(b)?(a):(b))

 

In file included from sketch\settings.h:34:0,

                 from sketch\limits.cpp:5:

sketch\nuts_bolts.h:31:0: note: this is the location of the previous definition

 #define min(a,b) (((a) < (b)) ? (a) : (b))

 
[... 282 lines cut ...]
avrdude: ser_send(): write error: sorry no info avail
avrdude: stk500_send(): failed to send command to serial port
avrdude: ser_recv(): read error: ��leyici ge�ersiz.

avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_send(): write error: sorry no info avail
avrdude: stk500_send(): failed to send command to serial port
avrdude: ser_recv(): read error: ��leyici ge�ersiz.

avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_send(): write error: sorry no info avail
avrdude: stk500_send(): failed to send command to serial port
avrdude: ser_recv(): read error: ��leyici ge�ersiz.

avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: ser_send(): write error: sorry no info avail
avrdude: stk500_send(): failed to send command to serial port
avrdude: ser_recv(): read error: ��leyici ge�ersiz.

avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

ThomasHeb, 2020-12-05:
ok, perfect. I know that error. Please give me some time for testing and working out a solution for you. I am currently a bit busy with an other project.

ThomasHeb, 2020-12-06:
Hi Ugur,

the warnings regarding redefinition are correct. I use some parts of arduino.h and spi.h but use an own definition of min/max/... (I see this warnings in my own compiler runs, too)

Wat confuses me a bit is the warning
ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
e.g. report_status_message(gc_execute_line("G1X0Y0U0Z0"));
Normale the char* is generated in RAM, when "G1X0Y0U0Z0" is loaded from ROM.
I don't see any of this warnings in my compiler run. But they should not led to a dark display, because these lines are only executed if you do positioning/homing/slicing.

Your Code size is different than mine:
ROM 65672 vs 65992 (yours)
RAM 5817 vs 5916 (yours)

We had some problems with dark displays in past, when the libraries have not been installed correctly (or different versions). can you please remove your local libraries from your local arduino (sketch) folder
[image]
Please remove the U8g2 and SDFat_-_Adafruit_Fork. and all similar libraries somehow relating to U8-Display and SDFat, by deleting the complete folder (or backup it somewhere far away).

Please copy from your GitHub download the 4AxisFoamCutter-master/03_Firmware/libraries/
the two folders U8g2 and sdFat_-_Adafruit_Fork to your local libraries from your local arduino (sketch) folder.

Close and reopen the arduino IDE and try to compile and download again.
Please give me your RAM/ROM size. I am using Arduino 1.8.10 on a Mac.

worst case, I can generate the hex files for you... just let me know which hardware are you planning to use (with buttons or without; with or without limit switches).

Regards
Thomas

ugurdem19, 2020-12-08:
I deleted the library and installed different versions, but still get the same error. I don't want buttons and limit switches. I would be glad if you set it that way.

ThomasHeb, 2020-12-10:
Hello Ugur,

sorry for my late reply, I was busy the last days.

You must copy the libraries from the GitHub download 4AxisFoamCutter-master/03_Firmware/libraries/ to your Arduino/libraries path, with is located on My Documents on Windows systems.

I did a test with a Windows system and Arduino 1.8.11. The code is working, but the power of the display after reconnecting to USB Power took 10 Seconds.... very strange.
With Arduino 1.8.10 on Mac, the display is available 1 or 2 seconds after reconnecting to USB Power....
Very Strange, maybe you can load a 1.8.10 version and test.

Appending you find the hex file for operation without buttons and without limit switches
[FoamCutter.ino.mega.hex.zip](https://github.com/ThomasHeb/4AxisFoamCutter/files/5672456/FoamCutter.ino.mega.hex.zip)

I would recommend, that you install at least limitswitches close to your 0;0/0;0 coordinates, so that you can perform a homing... automated traveling of all axis to that limit switches and then a defined bit forward to a defined zero-position.

ugurdem19, 2020-12-11:
I did everything you said before, I tried with different versions, I tried copying and replacing the library files you provided, all of them got an error code. Thank you very much for your interest. you are a very good person.

ThomasHeb, 2020-12-11:
Hope you can work with the hex-file. If you need any changes let me know, I can provide updates.

Regards
Thomas

ugurdem19, 2020-12-12:
I installed hex I have not installed the motors yet, after I tried it with the motor, I will install the old foam cutting cnc. How can I do motor step adjustment in the menu.
Thank you very much for your help. A great work. Also, how to convert gmfc files to dat file x y u z gmfc is very easy for me. I don't know jedi.

ThomasHeb, 2020-12-12:
Hi Ugur,

for changing the parameters inside the Arduino Firmware (grbl) you connect the Arduino Board to the USB. open Arduino IDE and open the Serial Monitor from the Arduino IDE/Tools menu.
Select a baud rate of 115200.
With the command $$ you can access the settings menu. to change a parameter simply type $x=y with x is the number of the parameter and y is the new value
[image]
see this video https://www.youtube.com/watch?v=Bbq8E3omgh8 or search for "grbl change settings"

I don't have any plan on GMFC. I use Winghelper for my wings and SketchUp for my fuselage. Currently I am working on a SketchUp Plugin to generate more complex gcode, including post processing. Sketchup is for free in an older version.

Please make a simple wing design, some screen shots and send me the GMFC file, or even better, check if you can find a export or gcode export and send me this file.

ugurdem19, 2020-12-12:
Thank you very much for your help, you thought everything very carefully. Do you have to work for rotary cutting?

ugurdem19, 2020-12-16:
[image]
I tried to install it again, it worked without any problems, I started translating the menu into Turkish. Thank you for your help

ThomasHeb, 2020-12-16:
Hello Ugur,

i have no background in rotary cutting or milling. I have some programming and hardware skills and started my career in research and development....  long time ago ;)

I am very happy, that you solved the problem. for translation you need to work in lcd.cpp only (and report.cpp only for parameter and messages on usb interface). 

Please keep me updated on your project and let me know, if you need further support.

By the way, if you are still searching a software for drawing shapes and calculating the gcode, have a look at this topic https://github.com/ThomasHeb/4AxisFoamCutter/issues/9
You find a rb-skript in that post, which works for Sketchup and generated gcode directly. I am still working on some details and the video.... hope I get this ready over x-mas. 

Regards 
Thomas

ugurdem19, 2020-12-16:
https://www.youtube.com/watch?v=uWO8kMGxqGU
The other day I saw a video like this but I uploaded the dat file but it failed.

ThomasHeb, 2020-12-18:
This is a really cool Java program. it works quite well. 
inside the gcode you should add some delays/pause for the hotwire to heat up.
At the beginning, change from 
[image]
to:
[image]
You can do this with a text editor. Change the file extension to .gcode and you can use it.
A .dat-file is attached. check your the forum https://www.rcgroups.com/forums/showthread.php?3467209-WingWire-4-axis-gcode-freeware/page3
 

Another cool tool is here... online version
http://www.diyrcwings.com/app/
this guy made a perfect job. you only need to rename the exported file to .gcode extension
[image]
[Seling.dat.txt](https://github.com/ThomasHeb/4AxisFoamCutter/files/5717423/Seling.dat.txt)

closed 2021-02-03 by ThomasHeb