ThomasHeb/4AxisFoamCutter #15: Help with Power Supply Choice

Kwadology, 2021-03-24:
Hi Thomas,

I've decided to build a 4axis foam cutter based on your project. I'm pretty handy at mechanical design but my electronics knowledge is lacking, (Limited mainly to building drones and planes) I've watched both your videos and RCKeiths, I have all of the electronic components that you mention apart from a power supply. I want to go the same path as you and use a single power supply, (I have a step down puck already). However, there is one element I don't quite understand before purchasing one. I know you are using a Lab supply and that the hotwire is powered via the Ramps 1.4 board and you can control the power output through your controls / software via PWM. My questions is where is the PWM signal being sent to? Does your Lab Supply have a PWM in to allow it to be controlled via this method or is the Voltage / Current from the lap supply constant and the Ramps board itself has the ability to control the power being output to the hotwire? (Presumably in that case up to a Max of whatever the lab supply is set to). 
None of the power supplies I've looked at seem to have any kind of PWM input port that I can see. I'm looking at this one: https://www.amazon.co.uk/Variable-Eventek-Adjustable-Regulated-Alligator/dp/B07DQQBJF6/ref=sr_1_34?dchild=1&keywords=Variable+Power+Supply+30v+10A&qid=1616586189&sr=8-34
However if the power is controlled by the Ramps board is there any advantage of having a variable power supply at all and would it be better to choose something like this https://www.amazon.co.uk/Surom-Universal-Regulated-Switching-Computer/dp/B071W121H7/ref=sr_1_51?dchild=1&keywords=Power+Supply+30v+10A&qid=1616587654&sr=8-51
I'm planning on making quite a large cutter with probably 100cm - 120cm hotwire and will want to cut EPP foam so conscious I may need more power than cutting other types of foam, so want to make sure I have some headroom. Thanks in advance for any help / advice you can give.

SimonJohnCastle, 2021-03-24:
Interesting Question.
In my case I went for a 60V 5A bench supply as you need more voltage for longer wire lengths and I wanted to be sure.
If the output from the lab supply is fed through a PWM switching FET etc to further control the average power you can lock the
PSU into the wrong voltage and current limit, just to be aware if you observe this.

Kwadology, 2021-03-24:
Thanks very much for your reply, I wasn't aware about more voltage for longer length that's really useful. I'll have a look at some calculations to make sure I know roughly what I'll need. 
So is there a pwm switching FET on the Ramps 1.4 board  or are you saying if I add one in somewhere? (I dont remember seeing this mentioned in the components list). I dont really see a need to be able to control the power through the Arduino control if I can do it through the lab supply, but conscious that if I'm using one power supply to power the hotwire with a step-down puck in parallel to power the Arduino, ramps, steppers etc. it will consume some of the power that i'll need to account for. 
I guess what I was originally trying to ask as well, is that if the ramps has a switching get capable of controlling the power is there still a benefit of using a variable Lab style power supply like the one you mentioned vs a fixed output PSU?

SimonJohnCastle, 2021-03-24:
There is one on the ramps board, but it looks like its limited to 30V or 35V, I think Thomas mentions this some where.
You can always use an external FET if need be but then controlling the power needed at the right time in GCODE may be a bit tricky.
Better to just use the FET or Relay to simply turn the power on FULL and adjust it by hand on the PSU as needed.
You generally need to turn it up a bit a second or two once it enters the foam, to start with its too hot, then it cools in the foam, then you need more power

Kwadology, 2021-03-24:
Great, thanks for taking time to help with this. It sounds like 2 power supplies and controlling the hotwire power manually is the simplist way to go. Ill check out the power calculations for the wire and go from there.

ThomasHeb, 2021-03-24:
Hi Guys,

Thank you for sharing your knowledge and supporting each another. I would like to add some more information and offer further support.

The Ramps has a MOS FET on Board. When you use a Ramps Board 1.4 the MOSFET for the hotwire has an independent power connector. This means you can use one voltage level for the stepper and the Arduino and an other level for the hotwire. I strongly recommend to use a Ramps 1.4 Board. Using a Ramps 1.6 Board allows only one voltage level and you are limited to the maximum voltage for the Arduino (12V... max 17V) and you need do play with the thickness of your wire..... that's not the best way, because you want to keep the wire as thin as possible.

The on board MOS FET can operate up to 60V DC. The MOS FET is used to switch the hotwire on and off and to drive a PWM. I wrote up to 35V in the hardware section, but for the hotwire MOS FET, you can operate up to 60V DC. There are no limiting parts connected. I guess the 35V is coming from any capacitor limitation .... but not inside the hotwire part ... and sorry, I do not find the exact reason why I wrote 35V.... maybe too conservative. 

Simon is on the right way. More voltage keeps the current low and the hotwire thin and allows longer hotwire length.
I am using a similar power supply with up to 30V... that fits perfect to my wire length.
have a look at https://github.com/ThomasHeb/4AxisFoamCutter/issues/5
PWM is finally only a kind of fine tune.... 

If you tell me the operation width of your machine, I can do a quick calculation for a thin wire. Best would be the length of the wire, when it is fully diagonal from lower left front to upper right back corner.

Using a buck converter (step down) is a kind of standard for my hardware - but not required. I want only one laboratory power supply... so i need to generate the 12V for Arduino/Stepper out of this supply, too. But It works fine, too, if you use a second power supply.

Best regards,
Thomas

Kwadology, 2021-03-26:
Hi Thomas, 
Thanks so much for your comments that helps a lot, sorry its taken time to reply ive been deciding on final dimensions.  The hotwire will be 120cm, maybe 125cm between connections. (Max diagonal length). 
I would definitely prefer to use only 1 power supply, I have a Ramps 1.4 board and already have a buck convertor although its only rated for 40v max so I may need another if I should choose a 60v rather than 30v power supply. 
[£3.70 62% OFF]Geekcreit® DC 6-40V To 1.2-36V 300W 20A Constant Current Adjustable Buck Converter Step Down Module Board Module Board from Electronic Components & Supplies on banggood
https://banggood.app.link/5c1j5t8kWeb
If your able to suggest what wire I should start with and if 30v is okay or I should choose 60v, that would be an amazing help. I do want to cut epp foam which I know is pretty dense so ill definitely need a slow feed rate but I'm not sure if I'll also need more power. Thank you.

ThomasHeb, 2021-03-26:
Hi,

slow feed rate is always better than more power, because you must keep in mind, that one axis may travel slower than the other one, so higher power may case unwanted melts... have a look at the speed correction and the tower example video.
In this video, you can see how to setup the hotwire power for low cutting speed.
https://github.com/ThomasHeb/4AxisFoamCutter/files/5382446/IMG_4588.mov.zip

I would recommend this wire (I use it). It is thin 0,2mm, app 14 Ohm/foot or 46 Ohm/m. I am cutting XPS Foam... the yellow one from the video, this is similar to EPP.
https://www.amazon.de/Proxxon-28080-Ersatzschneiddraht-Thermocut-Rolle/dp/B0016ODWWC/ref=sr_1_5?__mk_de_DE=ÅMÅŽÕÑ&dchild=1&keywords=proxxon+heißdraht&qid=1616770703&s=diy&sr=1-5

Appending the calculation. I recommend a 60V DC Supply, because you need 42V minimum.
For normal operation I would start with 50V and adjust the hotwire power setting (see video above).
[image]

The Geekcreit® DC 6-40V won't work. you need to find a DC DC step down with input voltage up to 60V DC or take a separate supply for the arduino/stepper motors (output 12V DC... app 3 A rating. You will need app 1A in normal operation, enough reserve to work without cooling/fan for the supply).

Best Regards,
Thomas

Kwadology, 2021-03-26:
That's an amazing help, thank you so much. I'll order  a 60v lab supply and look for an appropriate step down buck convertor. I'm sure I'll have questions as I work through but your comments in issue 5 are also really helpful. Thanks again.

Kwadology, 2021-03-26:
Hi Thomas, sorry for another question.  Could I just check this is a suitable buck convertor. Maybe a silly question it says 60v to 12v up to 10A. I'm just surprised there is no heatsinks on it like the lower voltage ones.
https://www.amazon.co.uk/dp/B07DNHXWV5/ref=cm_sw_r_cp_apa_fabc_AA607BZ570A61PDADSBH?_encoding=UTF8&psc=1
Thanks

Kwadology, 2021-03-26:
Actually the reviews for that look bad. Maybe with the higher input voltage a 2nd power supply is a better idea.

ThomasHeb, 2021-03-27:
Hi,

the module should work, but you are right, the reviews do not look good.
I checked out some modules and it is hard to find something above 50V. I think the main limitations are capacitor voltage rating, linear regulator for chip supply and voltage rating of MOSFEET. If you have good knowledge in electronics, you can upgrade the parts. Otherwise I would recommend to buy a external supply.

Kwadology, 2021-03-28:
Thanks for highlighting that psu. Much cheaper than I thought. Thanks again for your help.

SimonJohnCastle, 2021-03-29:
I guess, you have solution by now, but I if not you could always use a boost converter for the high voltage.

closed 2021-04-03 by ThomasHeb