ThomasHeb/4AxisFoamCutter #17: Some Questions by Neo

ThomasHeb, 2021-08-05:
Request from youtube 
https://www.youtube.com/watch?v=WJ8nHyDqIhs&lc=Ugz1C6c4nUitd0avGOt4AaABAg
by Neo2001cro

I will try to break the questions in some related topics and explain them step by step. Please feel free to ask for further details

===================

Thomas thank you.
Your 'work', sharing knowledge, plans, pictures, scripts... is apsolute great, excelent! Respect. Thank you.

And if i can, have you 'dialed in' your Foam Cutter to the German precision ;)?
Do you have wire temperature under control, are the tolerances aceptable?
Can you please show some of the 'action' of the Foam Cutter and Laser Cutter? How many models have you make with it?
Your foam cutter can make shapes in what dimensions, for example, can your foam cutter make DLG wing, 750mm long, with 50mm (170mm)  airofil on one side, and the 25mm (90mm) on the other? (becose of the wire 'pitch', is there enough height in your foam cutter) ?

And, you have shared plans, but, wire tension sistem, holes in aluminium profiles, brackets... Can you please explain, show that little bit more, like, idiot proof ;) ?
Steper drivers, i bet that drivers works great, super, but ;) if you listen them for long time they become unpleasant, or am i wrong? Have you try, do you recomend some quieter drivers? 
Ramps 1.6 is better then 1.4?...

Your laser cutter looking great to, but, please share some more videos, what have you succed to done, what have you not...

Again thank you and please make some more videos, upgrades... you are on the right path, respect. 
IMHO your Foam Cutter and the Laser Cutter are on the right 'track' to become No.1 for the RC people! Please don't stop perfecting it.

I am sorry for my very bad eng :(

ThomasHeb, 2021-08-05:
**Foamcutter: Wire temperature / Dimensions:**

You can build the cutter as big or as small as you want. I have chosen the V2020 rails with NEMA stepper, because they are available on Banggood.com and only some modifications on the idler pulley are required. But they are oversized for that task. Google for rcKeith, this guy has a much more simpler mechanical construction.

What you should keep in mind:
X travel dimensions should be bigger than your wing depth: 170mm wing depth >>> min. 250mm operation depth
Y travel dimensions should be bigger than your wing height: 50mm wing height >>> min. 100mm operation height
because you will always need some foam surrounding your wing.

Wing span is related to the distance between the two axis pairs.
On one hand you should have some additional space...i.e 750mm (total span is 1,5m?!?) your operational distance should be 850mm. BUT if you have non parallel traveling (i.e. reducing wing depth on the outer tip), a big distance between the axis pairs will increase X and Y required. See pictures. 
I would recommend, that you build the axis pairs moveable, by clamping them on a table, or adjusting them along a rail. My solution is not a good one!

https://www.youtube.com/watch?v=WJ8nHyDqIhs&lc=Ugz1C6c4nUitd0avGOt4AaABAg

ThomasHeb, 2021-08-05:
**Foamcutter: Wire temperature / accuracy / tolerances**

Most important thing is, that your axis are orthogonal.... 90°. Assuming this is mechanically adjusted, the tolerance is related to:
- tolerance in the mechanical travel system (belts,...) and the flexibility of this system (more flex, more worth). Cutting foam does not need much power/force... so belt driven systems are ok. If I really put pressure on my system, I can move it about 1mm.... but this is not realistic during cutting. it is important, that you reach the same point during back travel. try a screw to the right and then to the left, you will find some degree of turning, without in/out movement. screws can operate a lot of force, but get expensive when reducing this issue. I use a belt, because that issue is very small, if you tighten the belt. Belts can not operate a lot of force....but I don't need it ;o)
- Firmware. I use 80 micro steps per mm, so my max resolution is 0,0125mm... you need a high resolution to accelerate over the steps. so firmware is very accurate
- hotwire itself: if you melt a foam with a hotwire, you will see at least 0,5mm melting around the hotwire.... and THIS drives the accuracy! in Addition the thickness of the hotwire... use a thin one!

find the lowest temperature to cut the hotwire, by butting some foam on the hotwire and increase the power setting. see video in zip. Then do some test cuts with that power setting at certain speeds.... I do my cuts with 100mm/minute
[IMG_4588.mov.zip](https://github.com/ThomasHeb/4AxisFoamCutter/files/6937175/IMG_4588.mov.zip)

neo2001cro, 2021-08-05:
ThomasHeb thank you!
You are realy professional 

From where to begin ;)
I have wish, desire to build that cutter, but, i wanna that is very precise, and that meets every RC builder dreams ;)
No presure at all on you ThomasHeb ;) :)
You say 1,5m!, yes, true, its big, but ThomasHeb, if you can make foam cutter that can 'perfectly' cut, in very tight tolerances, DLG wing, 2x750mm long, maybe even in one go, left and right like on "WingWire", you will literally become RC God! 
You, and others will say that I'm exaggerating but belive me, that 'war', to simple build perfect DLG wing now is almost 30 years old :( (maybe even more).

ThomasHeb please make that true, become reallity.

DLG builder have become to that 'magic' number from years of experience (if i understod good), this 1,5m is excelent in air, and still easy to 'travel' with.
Now days, in 2021 there is a lot of 1m DLG that are really good, but, if you can cut 1,5m then you can cut it and 1m :)

You say that you see that "with sistem" on your cutter is not good, have you find better system?
Is, maybe, "DoDo Do" 'desing' is better, easyer, smarter... way to build?

And, i know, its much work, but what you tell to build somekind of "Parts list", that even people like me can, buy it, and build it, without much 'fuss'.

You say that combination of power, wire, 'components'... will always have some tollerances, so, can you make in your sketchup-cutter-tool-plugin option that will calculate, cut foam, in for example in 102% or even 98% of exact 'plans', like on 3d printer?

ThomasHeb, 2021-08-05:
Hello Neo,

Building a DLG is a combination of many tasks, like airfoil design, shape, weight, foam cutting, CFK, GFK, vacuum molding,.... and each of them on a very high level! and you need to understand the processes behind.

I can not give you a ready to rebuild part list for a cutter, which meets that accuracy without the technical understanding.

Example:
I have been measuring the cutting accuracy (0,6mm) and I add this in my SketchUp designs manually. 
If I need to cut slower or faster, I know how much I have to add or subtract from my shape. This can be calculated in a software or a sketchup script, but this needs weeks or month of work and testing.  This is not worth for the 5 planes I want to build and I can not spend this effort on a hobby. I use standard functions in SketchUp instead, but I need to decide manually, If I need an accurate outer or inner shape is needed... Scaling a wing by 102% makes holes to 102%, too.... but 98% are required for holes.

If you are looking for professional tools you need to buy on industrial standard level. Way to expensive for me and my 5 planes.

WingWire or other tools generating or sending gcode will work with my machine, too. Never the less, from design to real wing are some big steps. Even if you look at the videos for wing wire [link](https://www.youtube.com/watch?v=uWO8kMGxqGU) you see, that the wire is melting a big portion of the foam. You need to find this issues and work through the optimisation process. 

Regards 
Thomas

ThomasHeb, 2021-08-05:
In addition a Lo100 made with the FoamCutter
If you look at the tailbone, you see some waves on the surface. It took me three attempts to get to this quality, but will need a lot more to get to a smooth surface. I sanded the surface, covered it with two layers GFK and usesd some filler... this is a no-go for weightoptimzed DLGs.

[image]
[image]
[image]
[image]
[image]

ThomasHeb, 2021-08-05:
Hi Neo,

sorry to disappoint you, the formula is much more complex and needs to do some shape and travel decisions, ...
But please keep in mind, that developing a good DLG ist much more than foam cutting. You need to know a lot about airfoils, geometrical or aerodynamical shrinking, aerodynamical effects, V-wing, arrowing, rudder primary and secondary effects, stability calculation .... 
If you have a good design and if you understand what happens during normal flight from a theoretical  perspective, then generating the SketchUp model, adding the additional things for the cutter manually (wire diameter, ...) is less than 10% work. 

If you have such a design in electronic format, you can attach it here as zip and I can check is the cutter can produce it.

neo2001cro, 2021-08-05:
eh, i didn't expect that, that i will be responsable that you from Foam Cutter will get to developing DLG ;)
ThomasHeb i know everything what you try to tell me, but, everything in DLG 'starts' with 'perfect' foam wing, then come the 'rest'.

neo2001cro, 2021-08-05:
to be clear, i didn't draw that, its not mine, i 'find' it on the 'forum' ( i don't know if i can tell name)
i have it in .stl file, i don't know how to convert it to sketchup 2017 file :(

[DLG wing.zip](https://github.com/ThomasHeb/4AxisFoamCutter/files/6941970/DLG.wing.zip)

ThomasHeb, 2021-08-06:
Hello Neo,

the ramps 1.6+ works, but you have only one input voltage. For the linear 5V regulator you should stay below 17V. On the other hand, for long and thin hot wires you need a high voltage for the hotwire (30V + ). Thats why I recommend the ramps 1.4 with two independent input voltages. 
You do not need fans, so the only modifications are 100nF capacitors on the limit switches to reduce EMC.

In addition you need a Arduino and a display…. so it is better to go for a bundle
https://www.aliexpress.com/item/1005001631916842.html?spm=a2g0o.productlist.0.0.3a6055dc0GjnZ4&algo_pvid=ba86b7bf-b93b-4c9a-8be2-e77d3a6fbe9a&algo_exp_id=ba86b7bf-b93b-4c9a-8be2-e77d3a6fbe9a-26

If you do not need a display and stand alone operation without PC, you can go for a open source firmware with an easier hardware setup. i.e MKS Board. 
https://rckeith.co.uk/use-the-mks-gen-l-board-for-cnc-foam-cutting/
using grbl MeEGA 5X
https://rckeith.co.uk/grbl-hotwire-mega-5x-for-cnc-foam-cutters/

On the pages of Keith, you find a lot of very good documented setups for mechanics, electronics, firmware and software.

ThomasHeb, 2021-08-06:
Hello Neo,

you can import the str file in Sketchup in the menu file/import.
The wing is a complex one with geometrical and aerodynamical shrinking and well curved leading edge, looks like a competition one ;o)

Missing or open items for cutting foam:
- v-angle, all sections should be cut out with that angle, so that both wings can be glued together inside their moldes
- position for main spar, a 0,5mm spacing should support the rowings
- all airfoils / rips have a very low resolution.... corners, not smooth... so it will be hard to identify the airfoil type and update it
- the wing needs to be cut in sections, due to the curving of the leading edge. Width of the machine needs to be adjusted (reduced) for the wing tip sections
- All airfoils need to be stretched in depth. After cutting&glueing leading edge and rudders can be curved by sanding
- the airfoils are very thin.... that's the DLG idea ;o) so you will need to do several tests, to add the right spacing for the hotwire (thin foam tends to melt faster an do some strange bends).

I think I would need 
- 20 hours to update the wing model for first cutting
- 10 cutting attempts to optimise the model 
- I would cover this prototype with glas, sanding, filler, painting, to get the best surface I can. small errors can be repaired by adding epoxy/filler
- build a negative mold out of that prototype (epoxy/GFK/....
- for cutting the foam core for the wing, you can use the last optimised model, maybe you need to add some spacing for the spare, servos,...
- minor errors in the foam do not matter, because the outer shape is defined by the mold

Alternatively: 
- 20 hours to update the wing model for first cutting
- 10 cutting attempts to optimise the model 
- glue&sand it, prepare the leading edge
- you can laminate CFK/Epoxy directly in the foam-mold with a plastic spacer to get a good surface
- Jan-Henning did some great video tutorials (only german) https://foobla.wigbels.de/tag/jan-henning/ or https://www.youtube.com/watch?v=p5BsfmUrtDM&list=PLZdOy4nGe2QKGPChQ_nEaEKesgN3__jY1&index=1
- this method is best for small number of wings... more than three, I would go for the GFK-Mold

Building the hot-wire machine
- My mechanical design is very complex, check out rcKeith
- Unique is my hardware/electronics and firmware. you do not need a PC to control the CNC. Everything is executed from the SD card directly (no gcode sender needed)
- choose your design and order the parts... I did all over bang good.com 
- you will need some standard tools (drilling machine, thread cutter, ...)
- you will need app. one day to get the hardware/electronics running w/o connection to the mechanic ... motors turning (you need to know a bit about Arduino, electronics, soldering,..., using Arduino IDE serial monitor)
- you will need app. one week to build the machine and do some simple tests
- basic tests included (you need to know a bit about Sketchup, measuring,..)
- optimisation is done on the model and included in the cutting attempts

neo2001cro, 2021-08-06:
Hi ThomasHeb
"str" file? You mean .stl?
I don't see it on my version :(

[image]

Profesional wing, from where do you get that :P 
Why that i bother profesional with the beginer stuff :P
DLG guys cut that in 3 parts (for now, best way that i see is on hands with hand hot-wire, or very expensive CNC)

Gentlemen Jan-Henning i know ;) respect
For the gentlemen rcKeith i know to... he recomended you :P
IMHO your version of foam cutter is better...
I know little bit about Arduino, electronics, soldering...
But not even near like you do, **respect to you for your knowledge.** 

Thank you for your 'research' on how to make very good DLG, i know that alredy, but maybe someone doesn't.

You have share, give, realy good base, maybe even to good!

But i wanna that finaly can be something done for 100% that even my kid can do it alone.
So why to solder pins, if there is no need to do that (that why i have asked for 1.6+ ramps board)
I know that 1.6 is not good like 1.4, but, bigthree have 'mantioned' something about improving 1.6, so sory for 'trying', asking for  **ramps 1.6+**
bundle soluutions, why not, even simplier, thumbs up!

About airfoils, hehe that is my area ;)
I can give you airfoils with that many 'dots' that you will tell me that i am nuts ;)
You se, that not bothering me ;)

ThomasHeb, 2021-08-06:
Hi Neo,

that is strange, I see the STL in SketchUpMake 2017. Maybe go to the warehouse (menu window/extension warehouse) and search for a plug in.
[image]

I think, we are coming close to the point.
Hotwire CNC is a complete project, with required knowledge about mechanics, electronics and arduino.
To design an airfoil or wing for CNC, you need to know how to use SketchUp or a CAD tool and you have to adjust the design to your CNC.

I did one wing with foam core and GFK mold. It took me 10x longer doing the design in CNC, building the machine compared to cut the foam core in a traditional way with alloy-rips manually. So CNC is worth to go, if you see it as an own project, or if you want to reproduce the wings in a high volume. It is a long way to get CNC that running 100%,  you must decide if it is worth for you to spend the effort and the research.

neo2001cro, 2021-08-06:
strange (about .stl) ...
about time, shure, we all must deside is it worth our time and money.
But, your Foam Cutter looks good in 'action' so, why not build something so good!
If you can, ThomasHeb please make even some more improvements, like i say allready to many times ;) you are the closest that i have seen on being on, cross, finish line to make perfect RC Foam Cutter.

ThomasHeb, 2021-08-06:
Thanks, that you like my cutter.
I can support you, if you don't get something running. But actually I don't have the time to do bigger improvements. The weather is getting better, we got our Cessna 172 and start travelling around a bit and the RC gliders want to fly, too ;o)

Regards Thomas

neo2001cro, 2021-08-07:
> Thanks, that you like my cutter.
> I can support you, if you don't get something running. But actually I don't have the time to do bigger improvements. The weather is getting better, we got our Cessna 172 and start travelling around a bit and the RC gliders want to fly, too ;o)
> 
> Regards Thomas

Oh man, that is so beatiful to hear, that shold be, that people enyoj in such a gret hobby, not that always 'build' something....

But ;) :P 

When weather go bad, then, maybe ;) :P

Thank you THomasHeb for your work, its really really good!