ThomasHeb/4AxisFoamCutter #18: Update broke code...

LarsBrinck, 2023-01-23:
Hi Thomas, I´ve been cutting for some time with great result and your firmware has been working like a charm. Thanks! As I updated my setup with a new control-box I decided to implement limit switches and buttons as well (I´ve been using only the menu and SD-card/gcode to control the machine until now). When I updated the code - the machine went dead...

1. First I just took the old code and changed the "Set Limit Switch" and "Set Buttons" to 1. The code compiled and uploaded without errors but I got a blank screen. I checked the libraries (had been updated) and then reverted to the old libraries you have uploaded. Same result. No error but blank screen. I tried with and without the limitswitches plugged in and also with the -/GND switch inputs hard NC as I had my setup until now. Still just a blank screen. No sound from the stepper motors.
2. I then downloaded your updated firmware and got a compilation error: "DIOPIN_HOTWIRE_DDR" was not declared in this scope. I attach the full error code.

Do you have any clues or which way to go now? Are there settings in the new firmware code that I have missed? I use the Ramps 1.4 board. I don´t have any an or hotwire attached to the board.

[Error code Foamcutter 230123.txt](https://github.com/ThomasHeb/4AxisFoamCutter/files/10479292/Error.code.Foamcutter.230123.txt)

Cheers from freezing cold Sweden!

LarsBrinck, 2023-01-23:
As a last resort I rebooted my macbook and suddenly the code compiled and loaded. Very strange. Anyway, now it seems to work. Unfortunately I got so excited to test the limit switches so I started fiddling with my pliers in the Ramps board without disconnecting it from the mains. I shorted the + and - on the pins for the limitswitches. Something fried -f&#%k. Now the screen goes white when I run from the mains power. Powering through the USB still works so hopefully the arduino survived. 

Feel free to close this issue - If you have any link to debugging a fried Ramps board I´ll be more than happy. I will order a new one today...

ThomasHeb, 2023-01-23:
Hello Lars,

nice to hear that you are able to do some cuts with that project!

Updating the source is manly for the laser cutter ;o) But I know, that the library's always make some troubles. Inside the great display and SD library, some resources are used, which cause a conflict with the timing of the CNC/stepper processing... mainly timer resources.
So every update of Arduino IDE may also update the libraries, too. And Mac is doing that worth... don't ask me why, I love my mac, but the Arduino IDE seems to be more stable on the Windows PC.

So process the following way:
Download the firmware / library from [4 Axis Foam Cutter](https://github.com/ThomasHeb/4AxisFoamCutter)
Install (Copy!) the libraries from /03_Firmware/libraries directly in your ../Arduino/libraries/ folder (prefered, because updates of the original libraries may cause problems with grbl)
You may delete other versions of this libraries first
Select the version FOAM_CUTTER in config.h file
Do your Settings
Compile and download.

I shorted + / - for sensors on may CNC shield, too. In the end it was only a fuse.
Do you have the chance to measure voltage with a multimeter?
If so, please check:
- 1st page in attached pdf: ramps: blue dot = GND, red dots should have your input voltage for the steppers. Normally F1 is damaged.
- 5V for display,.... is generated from the arduiono mega board. So as long as you drive from USB, USB powers 5V level (3rd page, green areas, lower one is a comparator which always switches to USB 5V is available.)
- to check: Programm the blink example on your mega (disconnected  from ramps). it should blink the onboard LED. disconnect usb and supply over the DC connector on the arduino mega. it should blink. you should have your input voltage on GND/VIN on the POWER header. and you should have 5V on the GND/5V on the POWER Header. if not, you blow up the voltage regulator IC1 on the arduino mega (see page 4).

Regards
Thomas

P.S. I possible, send some pictures of your machine and projects.

[Hints.pdf](https://github.com/ThomasHeb/4AxisFoamCutter/files/10481915/Hints.pdf)

LarsBrinck, 2023-01-24:
Hi Thomas - thanks so much for your reply. I was not looking forward to 10h of googling on "how to troubleshoot a ramps board" so I ordered a new on the fly from my local vendor - but getting this troubleshooting guide for the setup is just awesome - I´m very grateful! 

So, analysis as per your instruction tells me that I´ve got 12.3V as input voltage on my steppers - does that mean that the Ramps board is ok? 

Checking mega - it loads blink, works on USB, when I give it 9V from a battery I read 8.6V at the DC-input, 7.6 at gnd/vin and 6.6 at gnd/5V... so that means that I blew the voltage regulator IC1 on the mega?

If so - can I replace the IC1 or do I need a new mega?

I zipped some pics of my very simple setup (https://www.thingiverse.com/thing:3676825) and some of the airframes I´ve been cutting. Mostly slope wings that I vac-bag but the last one with the psycedelic pattern is compression moulded with foam cut cores. Building and flying/thrashing slope "planks" is my addiction :)

[Exportbilder.zip](https://github.com/ThomasHeb/4AxisFoamCutter/files/10492861/Exportbilder.zip)

ThomasHeb, 2023-01-24:
**So, analysis as per your instruction tells me that I´ve got 12.3V as input voltage on my steppers - does that mean that the Ramps board is ok?**
This means the fuse and the diode is ok. there may be a problem with some capacitor, but I would say the ramps is ok….. maybe 
when powering the arduino with the USB cable and the ramps with 12V, can youmove the steppers within the menu POSITION? if yes, your ramps is ok.

**Checking mega - it loads blink, works on USB, when I give it 9V from a battery I read 8.6V at the DC-input, 7.6 at gnd/vin and 6.6 at gnd/5V... so that means that I blew the voltage regulator IC1 on the mega?**
his is a problem, the linear regulator must limit to 5V. you can change IC1, but there may be some other problems inside the voltage regulation part. I would propose to order a new mega board, too. Or get in touch with an electronic specialist, for checking all the components in around that area.

LarsBrinck, 2023-01-25:
I will borrow a new mega board today from a friend to see if it works. I will also try replacing the voltage regulator on the old one. If it doesn´t work I will just have to order a new one. Expensive hobby...

Thanks for all your help!

Regs,
/Lars

ThomasHeb, 2023-01-25:
I keep my fingers crossed, that only the regulator got hit.

Regards
Thomas

LarsBrinck, 2023-02-21:
Hi Thomas - got some time to dig into the foamcutter again and with the new borrowed mega and a new ramps-board it works again. All settings were gone but they were not that hard to restore in GRBL. So - doing job from SD-card works, homing works(!), so I can use the machine again. I´ve made some new wings  - tadaa!

Unfortunately there are some things that don´t work now and I would be happy if you could help me out:
1 - no buttons except the endless encoder works. Not even the "back" button on the display card. I´ve checked my new buttons and they short between ground and the different I/O pins according to the diagram in foamcutter.ino. USE_BUTTONS is set to 1
2 - the only menu-options that works are "Homing" and "SD-card". The other options does not work and since there is no "back" button you get stuck and have to reboot. In "Position" you can´t choose a position and go there - you can only choose "Set as home-position" and then you are stuck.

Is there any GRBL-setting that has to be set for the buttons to work. Do you think there can be something wrong with the display/reprap board? The "back" button itself works - I´ve checked it. And the display and encoder works.

Happy for all support!
/Lars

ThomasHeb, 2023-02-21:
Hi Lars,

with USE_Buttons = 1 the small knob on the display board is not used.

with USE_Buttons = 1, please boot to welcome screen, then connect D59 on ramps to gnd. this should move forward to main menue…. does that work?

Can you send me some pictures of the connection of the buttons on the ramps?

if you want, we can make a phone call tomorrow?
my email is thomas.heberlein@web.de, just send me your contact and a time proposal 

regards
Thomas

LarsBrinck, 2023-02-22:
Hi Thomas,
Thanks for helping! I´m no electronics guy so perhaps I´ve misunderstood the wiring. Here is a pic of my test (I just made a breadboard hookup) and a sketch of my wiring.

To get from bootscreen to mainscreen I can press the encoder. I see that D49 is occupied by the reprap-board. How do I technically attach anything there? Solder on top of the reprap-board?

Perhaps we can talk somewhere at 18.00-20.00 tonight? I´ll email....
/Lars

[image]
[image]

ThomasHeb, 2023-02-22:
Hi Lars,

time is perfect.
the D59 pin on the ramps is the back button. the firmware allows skipping the welcome screen with all buttons. so you can test the ramps/wiring. I will send you a image of the wiring in half an hour.
wiring principle as shown in your picture is perfect.

Regards
Thomas

ThomasHeb, 2023-02-22:
[RAMPS 1.4 (RepRap Arduino MEGA Pololu Shield).pdf](https://github.com/ThomasHeb/4AxisFoamCutter/files/10801100/RAMPS.1.4.RepRap.Arduino.MEGA.Pololu.Shield.pdf)

you can connect the buttons directly on the ramps... find three examples attached.
to test the buttons, boot the device and press the button in the welcome screen. if the button works (edge detection) it should skip to the main menu.

LarsBrinck, 2023-02-22:
Tadaa! I think I got the logic of the button implementation now - they work in the position menu as the encoder did before in "no buttons" mode. I think that the "back" button in D59 solved it. Thanks!
I´ll jump right in to printing a new case with buttons. No telephone call needed right now.

Thanks for helping me out again!
/Lars

ThomasHeb, 2023-03-04:
solved

closed 2023-03-04 by ThomasHeb